teodoryantcheff/SolidCam-Mach3-Post #1: Spindle start code, rotate direction

thuSkywalker, 2016-04-11:
The spindle start code is M4 when assigning CW tools to an operation and M3 when assigning CCW tools, while expecting M3 for CW tools and M4 for CCW tools. Haven't figured out where the problem is yet.

teodoryantcheff, 2016-04-11:
As per (this manual)[http://machmotion.com/documentation/Software/Mach4/Mach4-G-and-M-Code-Reference-Manual.pdf] M3 starts clockwise ("normal" direction) and M4 does CCW. Or I did not get your remark....

thuSkywalker, 2016-04-11:
@teodoryantcheff I meant that the gpp code is doing the contrary, see function @start_tool in all 3 gpp files (line 556)

teodoryantcheff, 2016-04-12:
:) Damn, your are right. Will fix it asap....

teodoryantcheff, 2016-04-12:
done. fixed in #73b89282453d308687d92b4fceaff22a651f1800

closed 2016-04-12 by teodoryantcheff