teodoryantcheff/SolidCam-Mach3-Post #3: Problem 3axis milling

Xenon7697, 2016-07-21:
I have problem with this post processor in simple 3axis milling. It is not mill as it need to mill it.
There is photo how it need to be milled. There is SolidCam simulation.
http://content6-foto.inbox.lv/albums/x/xenon25/Others/Capture3.jpg

And there is foto how Mach3 is milling in real.
http://content6-foto.inbox.lv/albums/x/xenon25/Others/Capture1.jpg

What is wrong?

teodoryantcheff, 2016-07-21:
Based on the [minimal] information provided I'd guess your Mach3 config is wrong  and you have IJ mode set to absolute and not incremental in general config.

Xenon7697, 2016-07-23:
You right I have not been set IJ Mode to incremental. Now I have done it but result is the same.

I have fix issue by reinstalling Mach3 to newer version v3.043.053 and now all is working!
Thank you for replay!

closed 2016-07-23 by teodoryantcheff