teodoryantcheff/SolidCam-Mach3-Post #4: G82 & G83 Drilling Operations

IdefixRC, 2017-01-26:
Hi guys,

first of all: AMAZING WORK and thanks a lot for doing this.
I have tested the post processor on a 3 Axis Machine and had some issues with the Spot and Deep drill operations.

Spot Drilling will output the following code:
( -------------------- )
( 4-SPOT DRILL - DRILL )
( -------------------- )
G0 X23. Y9.615 Z10. A0.
G82 Z-2.79 R0.21 P0 F200
  X40.2 Y9.615 Z0.21
  X57.4 Y9.615 Z0.21
  X74.6 Y9.615 Z0.21
  X91.8 Y9.615 Z0.21
  X109. Y9.615 Z0.21
G80
( ---- End of 4-Spot Drill ---- )

Mach 3 (at least mine) does the first hole as planned but then moves to hole 2 at height 0.21 only, does not spot drill and move on to the next holes.

Same effect with deep drilling:
( ------------------------- )
( 5-4.2MM-DEEPDRILL - DRILL )
( ------------------------- )
G0 X23. Y9.615 Z10. A0.
G83 Z-7.732 R0.21 Q0. P0 F200
  X40.2 Y9.615 Z0.21
  X57.4 Y9.615 Z0.21
  X74.6 Y9.615 Z0.21
  X91.8 Y9.615 Z0.21
  X109. Y9.615 Z0.21
G80
( ---- End of 5-4.2mm-DeepDrill ---- )

I manually edited the code to add the G83/G82 string after each re-positioning move and it works fine after that.

Did I do something wrong?

Thanks !

teodoryantcheff, 2017-01-31:
Hey. Thanks for the feedback. I'll to reproducing this soon and will come back to you.

teodoryantcheff, 2017-02-02:
So - I'm almost sure you are not running PostDevTestMachine. Do try that and let me know. Also which version of Mach.
PostDev is the post that I was using for the new implementation and need to properly rename it and delete it after so people do not get confused.

Cheers,

biqut2, 2017-11-16:
I experience the same issue. I have to uncheck the use drill cycle option to drill more than one hole per operation.

IdefixRC, 2017-11-17:
Using PostDev works perfectly for me. Thanks

teodoryantcheff, 2017-11-17:
@biqut2 did you try the postdev post ? 
I definitely need to make that the "default" so that people do not get confused with the older version.

biqut2, 2017-11-18:
I just looked an no I was not. I was using the Mach3_4X version. I did notice something that is consistent across both files and it is the following line: {nb, 'G0 G40 G49 G80 G21'} I always have to go in and edit this to changed the G21 to G20. Wouldn't it be better if this were not hardcoded like this?

teodoryantcheff, 2017-11-18:
Yeah, we need to make this dependent on the units used in the source part but since I'm a metric guy a have hardcoded that to G21. Will check what I can do about that.

teodoryantcheff, 2017-11-18:
@biqut2 bbe26ee6e24970cac9163c5f84275e207c8c5ab9 should be solving that.

teodoryantcheff, 2017-11-18:
Closing that issue. If G21/20 is still a problem please open a new ticket.

closed 2017-11-18 by teodoryantcheff