teodoryantcheff/SolidCam-Mach3-Post #5: Questions about usage.

rvega, 2017-04-11:
First of all. Thanks for sharing this :)
I have modeled a part in solidworks and I have access to a CNC machine controled by Mach3 software. See [this video](https://www.youtube.com/watch?v=7NTRUOdFMAU) for what I'm trying to do.
If I understand correctly. To use this postprocessor. I need to install it as instructed in the readme file, choose the Mach3 machine in solidcam and then export the Gcode.
Is there a way to simulate the code outside of solidcam, using the exported Gcode files, for verification before taking them to the machine?
Does the postprocessor support face and milling operations from solidcam? (those kind of operations are the ones I'm using)
Thanks in advance for any help! :)

rvega, 2017-04-11:
Also, when changing the machine of my cam part project, solidcam is asking me to convert some drill cycles types as seen in the following screenshots. Please advise.

[image]
[image]

rvega, 2017-04-11:
One more... What is the "trace level" asked when outputting the Gcode?

Thanks!

rvega, 2017-04-30:
I finally figured out things by following tutorials and experimenting... closing

closed 2017-04-30 by rvega

teodoryantcheff, 2017-05-02:
Been really busy as of lately, hence long time now reply... Let me know if you still have questions I will try to help.
X and Y versions of the post I made for machine that have their 4th, rotary (A) axis parallel to either the X or the Y axis of the machine.... I don't think this will make a difference to your usecase.