teodoryantcheff/SolidCam-Mach3-Post #10: No feed rates generated in G-code

BartBooz, 2018-06-12:
First of all great work! 
I'm using the "rotary maching 4 axis" option, unfortunately the generated G-code doesn't include the specified tool feed rates.

teodoryantcheff, 2018-07-02:
Can you elaborate, @BartBooz? My, and others' tests have been correct so far.

teodoryantcheff, 2019-01-21:
Closing this due to lack of feedback.

closed 2019-01-21 by teodoryantcheff

tsipa, 2023-08-07:
This might be usefull, but i personally have not idea how to read that ATM.

(1)@line_5x         ==> xpos:0.000F ypos:-1.408T zpos:55.832T feed:2400.000F 
                    ..> apos:670.991T bpos:0.000F 
                    ..> dapos:0.535 dbpos:0.000 
                    ..> xhpos:42.143T yhpos:-1.408T zhpos:36.623T 
                    ..> xpos_normal:0.7548086T ypos_normal:0.0000000F zpos_normal:0.6559451T 
                    ..> xmpos:0.000F ympos:-1.408T zmpos:55.832T 
                    ..> xopos:0.000F yopos:-1.408T zopos:55.832T 
                    ..> xtpos:0.000F ytpos:-1.408T ztpos:146.632T 
                    ..> xtopos:0.000F ytopos:-1.408T ztopos:146.632T 
                    ..> xrpos:0.000F yrpos:-1.408T zrpos:55.832T 
                    ..> xrtpos:0.000F yrtpos:-1.408T zrtpos:146.632T 
                    ..> dir5a:ccw dir5b:cw 
                    ..> x_surface_point:41.690T y_surface_point:-1.408T z_surface_point:36.229T 
                    ..> x_surface_normal:0.7548086T y_surface_normal:0.0000000T z_surface_normal:0.6559451T 
                    ..> feed_type:'feed_rate' 
                    ..> inverse_feed:0.000 feed_teeth:0.050F 
                      >  Y-1.408 Z55.832 A670.991 F0.

tsipa, 2023-08-07:
I tried disabling use_solidcam_feeds and got no F over entire file at all:

[image]

tsipa, 2023-08-07:
I did:
@line_5x
    gcode = 1
    if use_sc_feeds then 
        {nb, ['G'gcode], [' X'xpos], [' Y'ypos], [' Z'zpos], [' A'apos], [' F'feed] }
    else
	    if original_feed > 0 then
            {nb, ['G'gcode], [' X'xpos], [' Y'ypos], [' Z'zpos], [' A'apos], [' F'original_feed] }
	    else
		     {nb, ['G'gcode], [' X'xpos], [' Y'ypos], [' Z'zpos], [' A'apos], [' F'feed] }
	    endif
    endif
endp

and it's better now, but I don't fully understand wtf is original feed and if it's even a thing