teodoryantcheff/SolidCam-Mach3-Post #12: Explain use_solidcam_feeds

pszturmaj, 2019-06-10:
Could you please explain use_solidcam_feeds variable? I'm getting correct feeds for 4th axis when it's set to NO. Sometimes if I set it to YES, the 4th axis gets very high feed rate and I couldn't find the cause of this.

teodoryantcheff, 2019-06-10:
It's been quite some time since I last dealt with that, so you'll have to take a look at the "manual". IIRC it was a windows help file shoved somewhere under the SC directory tree.

pszturmaj, 2019-06-10:
I already checked those help files and found nothing. It seems that you introduced this variable yourself :)
https://github.com/teodoryantcheff/SolidCam-Mach3-Post/blob/master/Mach3_4X_Y.vmid#L344 - maybe it's something about 5 axis kinematics in SC?

teodoryantcheff, 2019-06-11:
I have no recollection whatsoever, to be honest... Maybe, as you say I did add this variable.

closed 2020-01-02 by teodoryantcheff