teodoryantcheff/SolidCam-Mach3-Post #13: 4 axis machine x

mwrtis86, 2020-01-02:
Hi..happy new year..!! Can you tell me please if you have post processor for 4axis machine  x..??

teodoryantcheff, 2020-01-02:
Hey, happy new year to you too!
As far as I remember both the along X and along Y posts are in. 
Am I wrong?

mwrtis86, 2020-01-02:
Thanks for the answer.. yes it is in old file but along x It doesn't show the displacement of the fourth axis to the code

Στάλθηκε από το iPhone μου

teodoryantcheff, 2020-01-02:
Well... It's been quite some time now. What I can suggest is to just swap your machine axes ... Or make a mach profile with x and y swapped. That should get you going.

mwrtis86, 2020-01-02:
Are you going to create a post processor for the x;

Στάλθηκε από το iPhone μου

> 2 Ιαν 2020, 13:04, ο χρήστης «teodoryantcheff <notifications@github.com>» έγραψε:
> 
> Well... It's been quite some time now. What I can suggest is to just swap your machine axes ... Or make a mach profile with x and y swapped. That should get you going.

teodoryantcheff, 2020-01-02:
> Are you going to create a post processor for the x; 

Most likely not. Not anytime soon for sure.
Would totally appreciate a PR though.

Cheers,

Teo

biqut2, 2020-01-02:
As far as I know last time I used this post processor it did support both x an y. In any case swapping out two letters is trivial.

teodoryantcheff, 2020-01-02:
There was something with `a` and `b` axes being the defaults along x and y... 
`angle_4x_around_y` and such...
Due to my lack of time now, I cannot sit down and play with that -- hence suggesting to @mwrtis86  to just swap his machine axes and go with that.

teodoryantcheff, 2020-01-06:
@mwrtis86 Unless you have further questions about that, I'm closing the issue.

teodoryantcheff, 2020-01-06:
@biqut2, @mwrtis86 is correct -- I redid the post back in time and now only the Y version is in.

mwrtis86, 2020-01-06:
Ok thank you

Αποκτήστε το Outlook για iOS<https://aka.ms/o0ukef>
________________________________
Από: teodoryantcheff <notifications@github.com>
Στάλθηκε: Monday, January 6, 2020 12:43:07 PM
Προς: teodoryantcheff/SolidCam-Mach3-Post <SolidCam-Mach3-Post@noreply.github.com>
Κοιν.: mwrtis86 <aslanis_gewrgios@hotmail.gr>; Mention <mention@noreply.github.com>
Θέμα: Re: [teodoryantcheff/SolidCam-Mach3-Post] 4 axis machine x (#13)

@biqut2<https://github.com/biqut2>, @mwrtis86<https://github.com/mwrtis86> is correct -- I redid the post back in time and now only the Y version is in.

closed 2020-01-06 by teodoryantcheff